pdxlocations/contact #3: error thrown when sending a DM or moving in UI not sure

SpudGunMan, 2024-06-20:
```
./curses-client.py 
Traceback (most recent call last):
  File "/home/spud/meshtastic-curses-client/./curses-client.py", line 445, in <module>
    curses.wrapper(main)
  File "/usr/lib/python3.10/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/home/spud/meshtastic-curses-client/./curses-client.py", line 384, in main
    select_channels(1)
  File "/home/spud/meshtastic-curses-client/./curses-client.py", line 289, in select_channels
    update_messages_window()
  File "/home/spud/meshtastic-curses-client/./curses-client.py", line 217, in update_messages_window
    messages_win.addstr(row, 1, wrapped_message, curses.color_pair(1) if prefix.startswith(">> Sent:") else curses.color_pair(2))
_curses.error: addwstr() returned ERR
```

pdxlocations, 2024-06-20:
Oh wow, someone is actually using this!

Hmm, I'm not able to reproduce this at the moment.  Most errors I've found are due to the window being too small when the client tries to draw and content doesn't fit.  Perhaps this is the case?

pdxlocations, 2025-01-21:
Closing this to make room for new bugs.  Happy to re-open if able to repro.

closed 2025-01-21 by pdxlocations