pdxlocations/contact #4: cannot import name 'module_config_pb2' from 'meshtastic'

SpudGunMan, 2024-07-09:
just updated to .14 python CLI and I suspect related? 

on cloning project and downloading .13 meshtastic python error seen 

`cannot import name 'module_config_pb2' from 'meshtastic'`

pdxlocations, 2024-07-09:
@ianmcorvidae do you know what the proper import would be now?

ianmcorvidae, 2024-07-09:
`from meshtastic.protobuf import module_config_pb2`

all the protobufs moved from `meshtastic` to `meshtastic.protobuf` in 2.3.13

ianmcorvidae, 2024-07-09:
You can also wrap it in a try/catch like this:

```python
try:
    from meshtastic.protobuf import module_config_pb2
except ImportError:
    from meshtastic import module_config_pb2
```

SpudGunMan, 2024-07-09:
> You can also wrap it in a try/catch like this:
```python
try:
    from meshtastic.protobuf import module_config_pb2
    from meshtastic import BROADCAST_NUM
except ImportError:
    from meshtastic import module_config_pb2, BROADCAST_NUM
```

settings.py also has problems with 
```python
from meshtastic import config_pb2, module_config_pb2, mesh_pb2, channel_pb2
```
not sure where it all is now (didnt look)

ianmcorvidae, 2024-07-09:
All of the `_pb2` modules are now in `meshtastic.protobuf` as of 2.3.13, so anything that imports those should have the try/catch too, probably

SpudGunMan, 2024-07-09:
if handy here is a fix https://github.com/pdxlocations/meshtastic-curses-client/pull/5

pdxlocations, 2024-07-10:
Closed by #5

closed 2024-07-10 by pdxlocations