pdxlocations/contact #6: Feature request UTF-8 coding

LOCOSP, 2024-08-08:
First of all thank You for such a great software 🫡. My native language contains many signs like "ąęźżłó" is it possible to make them work? Thank You in advance... In Screen shot should be written "chiałem żądać" and it is what it is now 😄 
[image]

pdxlocations, 2024-08-08:
I'm happy to hear you are enjoying it.  I don't know how to support other languages/characters but I will leave this open in hopes another developer can take it on.  Also, if I get more free time in the future, I will look into it myself.  May I ask what your native language is, and what country you are in?

pdxlocations, 2025-01-13:
@LOCOSP I've made a recent change that may fix this.  Are you able to give it a try?

LOCOSP, 2025-01-20:
I'm sorry to say that but it didn't work...

Just downloaded the newest 1.0.1

[image]

pdxlocations, 2025-01-21:
Thanks for checking.  Looking around I many be able to support this by adding:

```
import locale
locale.setlocale(locale.LC_ALL, "")
locale_encoding = locale.getpreferredencoding()
```

changing all addstr requests to, for example:
```
win.addstr(y, x, your_string.encode('utf-8').decode('utf-8'))
```

I'll give it a shot when time allows, or unless someone else does first.

closed 2025-01-21 by pdxlocations

pdxlocations, 2025-01-24:
Hey @LOCOSP, is it working for you now?

PR #43 (2025-01-21): Fix wide character input
Fixes #6