pdxlocations/contact #7: [FR] Single Window Module Support

infered5, 2024-08-14:
Add either a setting, launch argument, or separate curses-client program that has each individual 'windows' fullscreened in the terminal so that we reduce small screen crashes. Instead of having the node list on the right, channel list on the left, and messages dead center, we could make the entire window simply an entire module at a time. This benefits default terminal screens on smaller displays, as well as cyberdecks, cell phones or Palm Pilot like devices.

This would require a complete rearrangement of the program. The underlying functions remain the same, but window drawing will have to be redone entirely.

This also makes FRs like debug screen and packet viewer more easily addable, since placement considerations are not a factor in big screen mode. This means long shot or underutilized menus can be added willy nilly, such as Asciimap, debug menus, packet flow or even statistics graphs.

The hotkeys will have to be removed in favor of a single hotkey that brings you to the menu. Arrows will then allow you to navigate to Nodes, Channels, Settings, with Return being how you enter it. I believe Escape is a good enough universal "get to the menu" button that a hint at the bottom of the screen is unnecessary. 

After some brainstorming, I do not believe a pagination style ux is sufficient as users may become lost in what could be a very large carousel of options. I believe a single escape menu to navigate with arrows and return is more than sufficient, even if this project eventually balloons into having dozens of menu options.

pdxlocations, 2025-01-24:
Someday, it would be really great to see views like this.  Like a mobile/desktop switch that could be automatic based on screen size.

pdxlocations, 2025-06-07:
I just made the column widths configurable which gave me an idea.  If `single pane mode` is activated, we could just dynamically adjust the pane width and not draw the panes that aren't highlighted.  Might not need a huge restructuring after all...