pdxlocations/contact #23: Add support for sorting node list

rfschmid, 2025-01-14:
My node list is quite long. It's hard to find the ones I'm interested in. The list is just displayed in the order we get it from the node, which I think is just the order they were first seen in? It would be cool to have support for sorting like in the Android client:

- Last heard
- Alphabetical 
- Distance (? - with location fuzzing on default channels I personally care less about this one TBH)
- Hops away (I mostly use this to more easily see which nodes I'm talking to directly, so more a filter than a sort TBH)

pdxlocations, 2025-01-15:
Yes, this would be a great feature.  I'd like to keep the UI simplistic and I'm not sure if sorting would be a toggle, or automatic like how the mobile apps sort last heard?  I don't think distance and hops away need to be a priority. 

Of note: There's a nodeinfo list being stored in the sqlite db.  It's not used for anything yet.  I was thinking of pulling the node list in from there, but I'm not sure if that makes sense.  I figured there'd be a reason for it eventually. :-)

pfiliberti, 2025-01-23:
This is a good idea, I modified my list to show the short name after the long and the hops maybe if the tab key is used to advance to next window then left and right can be used to sort the list different ways? Ui's are tough.

closed 2025-02-02 by pdxlocations