pdxlocations/contact #28: Frequent Window Refreshes

pdxlocations, 2025-01-16:
Excessive window refreshing makes for a strobing experience on slower systems like the LuckFox

rfschmid, 2025-01-16:
Might not be the only cause but #24 probably isn't helping this, especially on app startup if the node has messages queued.

Ideas:
1. Check if the list order actually changed before triggering refresh
2. Identify which packet types can actually trigger last heard changes, limit processing to these
3. Implement a minimum time between list order updates
4. Update list order on a timer instead of in response to packet receipt (combine with 1)

pdxlocations, 2025-01-16:
#24 is one of the causes however, the windows also all refresh when navigating between windows and nodes unnecessarily.  The inverted items could be redrawn individually instead of refreshing the entire window.  I believe t'll take a bit of refactoring to correct.

rfschmid, 2025-01-23:
Maybe close? There are still some possible improvements but it's not nearly like it was.

pdxlocations, 2025-01-23:
I tried out the current master on a LuckFox and it's a waaay better experience.  The only strobing I see now is from:
Switching channels (and loading all the messages)
The settings dialog flickers when navigating. 

I'd say this is no longer a "bug" so I'll close.

closed 2025-01-23 by pdxlocations