pdxlocations/contact #35: Wrapped messages can crash draw_messages_window

rfschmid, 2025-01-17:
Logic for scrolling the messages window assumes each message is only one line when calculating the max messages/start index/etc for displaying messages. But we also wrap lines, so we can eventually attempt to draw outside the bounds of the messages window, leading to a crash with "addwstr() returned ERR". This only occurs if there are messages long enough to be wrapped and the chat is long enough to require scrolling the messages window.

Chats with traceroutes are especially likely to contain wrapped lines.

pfiliberti, 2025-01-21:
I had to change ui/curses_ui.py to get the msg window to scroll properly. In draw_messages_window() I changed max_messages calc to :  **max_messages = messages_win.getmaxyx()[0] - 4**. Not sure why **-4**. Also in adjusting the height if log is visible I changed max_messages to just subtract the packetlog_height. **max_messages -= packetlog_height**. This does not take into account the wrapped messages as explained above. and I use tmux for my terminal session but I don't think it makes a difference. Before these mods once the message screen filled up it would crash...

pfiliberti, 2025-01-21:
I fixed the above problem by copying the 2 lines that calculate the scoll position to do it again after the dynamic calc based on visible msgs and wraps and it corrected the issue.

pdxlocations, 2025-01-21:
I m ay have just gotten lucky with number of wrapped lines working out in #40 There are still issues with drawing messages and scrolling to the bottom of the message list

pdxlocations, 2025-01-22:
That sounds wonderful.

PR #40 (2025-01-21): fix text wrap line count
I had a couple of beers and a conversation with chatGPT and I think we fixed #35.  I'm not opposed to switching from win to pad but figured I'd try to solve it this way for now.
PR #36 (2025-01-23): Convert messages window to pad
Very much work in progress - @pdxlocations I was wondering if you'd be able to test this on the SBC you've got to see if this approach is feasible. I'm just making an arbitrarily large pad area and I dunno the performance implications of that. Scrolling the messages window this way however doesn't require fully re-drawing the whole thing so I'm wondering if it might actually perform better? I was thinking a good test could be to manually/script add a couple hundred messages into the DB, at least some of which wrap (but not so many messages that it would reach 1,0 [... cut]
closed 2025-01-23 by pdxlocations