pdxlocations/contact #65: Settings: longName / shortName change not saving

rfschmid, 2025-01-24:
* Settings > User Settings
* Set longName and shortName
* Save Changes
* Check name via official Meshtastic CLI client with --info flag
* Name is still what it was before, does not reflect change made in Contact UI

Device is Heltec V3. UI is otherwise working. Haven't tried much else in Settings to see if this is limited to longName/shortName/User Settings/etc.

Also tried rebooting through the settings menu after saving changes. Device successfully rebooted as commanded, but longName / shortName were not set. 

Also tried re-flashing device doing "Full Erase and Install" and re-loading my previous config via official Meshtastic CLI. Issue still occurs.

pdxlocations, 2025-01-24:
Tested on my end with a V3 with success.
- Make sure you're using the latest master, i broke and then fixed saving settings yesterday
- I assume you're hitting 'Save Changes' after entering a value.

rfschmid, 2025-01-24:
Pulled up to tips (92790ddca64b493328bb4e2c959fa02009b8f3c4) - still having problems. 

Checked client.log - every time I try to save changes here, I see: 

> 2025-01-24 10:53:33,297 - ERROR - Error saving changes: 'str' object cannot be interpreted as an integer
> 2025-01-24 10:53:33,298 - INFO - Changes Saved

Are we perhaps assuming that shortName should be interpretable as an int? Or maybe it's some other setting on my device that's tripping it up.

pdxlocations, 2025-01-24:
setting any one of long/short/licensed sets all three.  A guess it's the is_licensed bool maybe expecting a 1 or 0 but getting "True" or "False".  Odd thing is its working fine on my end.  I'll try a reset on my node.

pdxlocations, 2025-01-24:
I'm able to reproduce now.  I'll push a fix shortly.

pdxlocations, 2025-01-24:
Sort of a bandaid solution for now but should be fixed in 978e2942cbbba939a98fc0f05a54f5fe475d55e3

closed 2025-01-24 by pdxlocations