pdxlocations/contact #67: Feature Request

pfiliberti, 2025-01-24:
When your focused on the Channel Window, if you hit DEL key a window pops up asking for confirmation to drop the channel and then drops the database Table. Sometimes you add channels you didn't want by accidentally hitting return on the node window so this could allow removal..

rfschmid, 2025-01-24:
I was thinking about this too, but debating if it should drop the table totally or just mark it archived so if you started a chat with that node again it would preserve the previous history. FWIW, empty chats will disappear when restarting the app.

pfiliberti, 2025-01-24:
Maybe the box can ask the use remove or archive and act accordingly?

pdxlocations, 2025-01-24:
I definitely hope to include some database management like this.  Have the option to Delete OR Archive sounds good to me. We'd probably want to use Ctrl-D.

rfschmid, 2025-02-01:
I was trying to find the best location for a node earlier and now have a channels window full of nodes I only tracerouted and no longer care about so I might take a look at this next.

PR #110 (2025-02-03): Allow archiving chats
^d will remove a conversation from the channels list, but preserve it in the database in case we start a conversation with the same node again.

Did not implement deleting the chat from the DB entirely - can consider for future improvement, or sufficiently motivated users can delete the table from the DB manually.

Fixes #67
closed 2025-02-03 by pdxlocations