pdxlocations/contact #141: Mentions of "Femtofox" while not using Femotofox

rfschmid, 2025-03-18:
There are a few mentions of the "Femtofox" device in the help text for settings, which are displayed despite not using this client with a Femtofox device. Since this client isn't ONLY used for this device, I think we should use more generic language here - "Node" maybe? 

russ@russ-desktop:~/code/curses-client-for-meshtastic$ git grep -i "fox"
**localisations/en.ini:node_info_broadcast_secs**, "Nodeinfo broadcast interval", "This is the number of seconds between NodeInfo message broadcasts. Femtofox will still send nodeinfo in response to new nodes on the mesh."
**localisations/en.ini:hop_limit**, "Hop limit", "The maximum number of intermediate nodes between Femtofox and a node it is sending to. Does not impact received messages.\n[warning]Excessive hop limit increases congestion![/warning]\nMust be between 0-7."
**localisations/en.ini:admin_channel_enabled**, "Enable legacy admin channel", "If the node you Femtofox needs to administer or be administered by is running 2.4.x or earlier, you should set this to enabled. Requires a secondary channel named 'admin' be present on both nodes."

noon92, 2025-03-18:
Aha, good point. I just copied the original text I had in the hardcoded settings menu. Definitely remove references.

Sorry!

pdxlocations, 2025-03-25:
fixed in 3be31698dfcbdc6761cc1d1b7ccf2966f7d6acd2

closed 2025-03-25 by pdxlocations