pdxlocations/contact #142: Ignore Node support

rfschmid, 2025-03-20:
There's one particular spammer in my area that I'd like to not hear from (nor, ideally, relay packets for in the future.) 

The Android client seems to offer two different ways to do this:

- Tap the short name in a channel and check the "ignore" box
- Separately under Radio Configuration -> LoRa -> Ignore Incoming. 

The latter [is documented](https://meshtastic.org/docs/configuration/radio/lora/#ignore-incoming-array) as specifically being intended for testing purposes to simulate a node being out of range, and is limited to 3 node numbers, but has the advantage of explicitly having packets dropped (which is my ideal goal.) 

I'm not sure exactly what the "Ignore" checkbox when tapping a node does, but it doesn't seem to put the node number into the "Ignore Incoming" array for me [despite the documentation seeming to say that it should do that](https://meshtastic.org/docs/software/android/usage/#:~:text=or%20add%20it%20to%20your%20Ignore%20Incoming%20Array) (and I haven't spent much time testing this yet.) 

The contact client appears to offer neither option. I'm not sure why the ignore incoming array doesn't show up if it's part of the radio settings, but I don't see it.

pdxlocations, 2025-03-20:
The ignore-incoming array is deprecated and so we skip it when loading the menus:
https://github.com/pdxlocations/contact/blob/140d79421381c525d8598e61a2603d0c4b5155ab/ui/menus.py#L26

ignore is now added to the nodedb (like favorite) and neither has been implemented in contact yet.

pdxlocations, 2025-03-20:
These should be easy enough to add:
https://github.com/meshtastic/python/blob/5837bd0172a1761bc51e52bc7c03d3b0f17f3d34/meshtastic/__main__.py#L450-L474

rfschmid, 2025-03-20:
> The ignore-incoming array is deprecated and so we skip it when loading the menus

Just for my own education I'm curious where you found this information - I grepped through the meshtastic firmware and meshtastic python repos for "deprecated" and "ignore.*incoming" and I don't see anything about this at tips of main, and I haven't found it in the online docs either. I'll grant that blocking annoying people is clearly not the intended use of this option though :) 

I might try adding ignore / favorite node support when I get some time.

rfschmid, 2025-03-20:
Oh of course, documented in a discord message 😄

https://discord.com/channels/867578229534359593/919642584480112750/1329854463959109634

pdxlocations, 2025-03-20:
> Oh of course, documented in a discord message 😄

Haha, I think in the case of ignore_incoming Discord is indeed where I saw it.  Most deprecated fields are documented in the protobuf repo, for example:
https://github.com/meshtastic/protobufs/blob/14ec205865592fcfa798065bb001a549fc77b438/meshtastic/config.proto#L45

PR #147 (2025-04-02): Add Favorite / Ignore node support
Add favorite node support - still to do: Add some way to display which nodes are favorites, such as a star or heart next to the name. Fixes #128 . 

Add Ignore node support - maybe more importantly (since ignoring is more consequential than favoriting) still To Do is to provide some visual indication that a node is ignored. Fixes #142
closed 2025-04-02 by pdxlocations