pdxlocations/contact #155: If user enters an unset setting that is true/false, then exits without changing anything, that setting will be set to False

noon92, 2025-04-02:


rfschmid, 2025-04-02:
I think I can reproduce the described behavior. On my setup:

> Settings > Radio Settings > Deivce_Ui
> There are a bunch of settings labeled "Not Set" including "Alert enabled"
> Go to "Alert enabled"
> Press Escape or left arrow to leave without setting
> The setting is now "False" instead of "Not Set"
> Back out to the main settings screen
> Drill back down to Device_Ui again
> Setting still shows "False" instead of "Not Set"

Exiting and re-entering settings restores the setting to "Not Set", so this seems like just a display issue and not something where we're actually setting a setting that the user doesn't want to.

pdxlocations, 2025-04-06:
fixed in 031d74a2909ee6638a091fee61ddf2e2016c5b45

closed 2025-04-06 by pdxlocations