pdxlocations/contact #168: TypeError: unsupported operand type(s) for |: 'type' and 'types.GenericAlias'

jwarchol, 2025-04-16:
Installed 1.3.6 and it won't run, getting this error:

```
(meshtastic) joshua@MacBook-Air:meshtastic % contact            
Traceback (most recent call last):
  File "/Users/joshua/working/meshtastic/bin/contact", line 5, in <module>
    from contact.__main__ import start
  File "/Users/joshua/working/meshtastic/lib/python3.9/site-packages/contact/__main__.py", line 29, in <module>
    from contact.message_handlers.rx_handler import on_receive
  File "/Users/joshua/working/meshtastic/lib/python3.9/site-packages/contact/message_handlers/rx_handler.py", line 7, in <module>
    from contact.ui.contact_ui import (
  File "/Users/joshua/working/meshtastic/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 7, in <module>
    from contact.settings import settings_menu
  File "/Users/joshua/working/meshtastic/lib/python3.9/site-packages/contact/settings.py", line 12, in <module>
    from contact.ui.control_ui import set_region, settings_menu
  File "/Users/joshua/working/meshtastic/lib/python3.9/site-packages/contact/ui/control_ui.py", line 11, in <module>
    from contact.ui.menus import generate_menu_from_protobuf
  File "/Users/joshua/working/meshtastic/lib/python3.9/site-packages/contact/ui/menus.py", line 23, in <module>
    current_config: Message | dict[str, Any] | None = None
TypeError: unsupported operand type(s) for |: 'type' and 'types.GenericAlias'
```

Python 3.9.20

pdxlocations, 2025-04-16:
Thanks, it appears the syntax I use is only available in python 3.10 or higher.

jwarchol, 2025-04-16:
Hmm, makes me wonder why the heck I'm on 3.9 on this machine. I've got 3.13.2 installed on it in homebrew, but for some reason this venv is 3.9. I'll fix it, thanks for the heads up.

pdxlocations, 2025-04-17:
It wasn't too hard to support 3.9.  Fixed in #172

closed 2025-04-17 by pdxlocations