pdxlocations/contact #179: Noded/Node Info Window

pqhf5kd, 2025-05-27:
Currently the Node Info Window shows both the Short and Long name, while the Nodes Window shows the long name.
Similar to #153, is the Node Info truncated if the values are too long? (none of the nodes around me contain enough

Could the Node Window display the Short Name on smaller displays to free up 12 characters for the Messages Window?
Then could the Node Info Window display only the Long Name, freeing up 6 characters?

Could the Node Info Window be changed to a dialog?
Or a Node Info dialog in addition, which in the future could contain additional values?

pqhf5kd, 2025-05-27:
I don't have any particular requirements, if I do order some hardware for a dedicated device, I'll be sure to target 80 wide. 
I'm testing with an 80 wide terminal and the Nodes Window feels a little too wide relative to Messages and Channels so throwing some ideas out.

pdxlocations, 2025-06-07:
> Could the Node Window display the Short Name on smaller displays

Not really because if we haven't received a nodeinfo we don't know the short name