pdxlocations/contact #184: Feature Request: Metrics logging

rfschmid, 2025-05-30:
Mobile device clients have the capability of storing device metrics like battery level, channel/air utilization, signal metrics like SNR and RSSI for 0 hop nodes, position logs, etc. for both our connected node as well as remote ones we can see on the network. I'd like to see the ability to log that data in the Contact client as well. For example, I'd like to be able to see the historical battery level on my solar node, to see if my solar panel and battery are sufficient to keep the node up over time. 

Even if we just wrote this data out to a CSV file or similar, without the capability of displaying the history in-app, I think that would be a nice feature to have.

pdxlocations, 2025-05-30:
This is interesting and if I may think out loud:  In the beginning at least, I didn't want to log data since it could be considered personal information, especially position.  I'm more open to the idea of logging other metrics and thought about utilizing the sqlite db since it's already there but a csv could work too.

It would be pretty cool to display this data in a chart within the terminal window!  There are some projects out there that could be a starting point such as https://github.com/stephane-deraco/Termchart

Simplicity is a goal with this project, since this is a terminal client mostly meant for sbc's. I'm constantly on the fence about adding big features.  I also do my logging with Home Assistant, so this doesn't add value to me personally.  Meaning I probably won't do the work anytime soon but would certainly discuss it further and/or accept a PR.  I suppose just adding logging without visualizations would be easy, I may be inflating the FR more than I should... because I've thought about it previously.

[end thinking out loud]

rfschmid, 2025-05-30:
Thanks for the thoughts! 

I do have home assistant, but I haven't tried to connect any Meshtastic things to it. I'm curious if the logging you're able to get is just logs of a single node that's connected to the HA machine, or if it's able to also log metrics for other devices in the mesh? IE in my case, the HA machine is in the basement, the solar node is in the backyard and not within reliable bluetooth range, so could I get a battery graph of the solar node in this case? (HA Machine <-- Serial --> Meshtastic Radio <-- Mesh/LoRa --> Solar Node )?