pdxlocations/contact #195: [Feature request] Have a command parameter to enable verbose logs

Fabryz, 2025-06-10:
I'm trying to connect to a device via bluetooth, I use the `contact --ble Meshtastic_xxxx` command, the program starts up, but then crashes back to the terminal without any message.
It would be helpful to see why it closed up itsself, if there were any error.
Or to have a command option/parameter in order to enable this verbose debug log mode.

I have no issues connecting to my T-Echo, but I'm unable to connect to the Sensecap Indicator

pdxlocations, 2025-06-10:
The log is sent to a file in the contact directory.  I really should allow the connection errors to print to the terminal.  The reason you can't connect to the senscap is that the client API is in use by MUI.  You can only have one connection at a time.

rfschmid, 2025-06-10:
I usually have "tail -f contact/client.log" running in another terminal/pane.

Fabryz, 2025-06-11:
> The log is sent to a file in the contact directory. I really should allow the connection errors to print to the terminal. The reason you can't connect to the senscap is that the client API is in use by MUI. You can only have one connection at a time.

Forgive me, I don't understand
So if I turn on the Indicator, MUI loads up, no other devices connected via BT, I would never be able to connect with the contact command via BT? But only with the serial USB

pdxlocations, 2025-06-12:
I honestly never use bluetooth on a desktop.  Does your device connect to the web client with ble?  Do you get any output with `meshtastic --ble-scan`?
I don't have an indicator but I was just now able to connect to my tdeck using `contact --ble 'xDEC_a158'`. It took a couple of tries.

pdxlocations, 2025-07-17:
I've made inition connection errors show in the console in #207 
Hopefully this helps diagnose the app not launching when the node isn't connected.