pdxlocations/contact #204: crash when selecting the MediumFast channel

alestanga, 2025-07-14:
I've been using contact for about a month now, I love it, but yesterday it started crashing every time I select the MediumFast
the error in the log file is the one below:

2025-07-14 10:12:38,401 - ERROR - Console output:

2025-07-14 10:12:38,403 - ERROR - Fatal error: embedded null character
2025-07-14 10:12:38,408 - ERROR - Traceback: Traceback (most recent call last):
  File "/home/radio/.local/lib/python3.9/site-packages/contact/__main__.py", line 128, in start
    curses.wrapper(main)
  File "/usr/lib/python3.9/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/__main__.py", line 105, in main
    main_ui(stdscr)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 119, in main_ui
    handle_up()
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 184, in handle_up
    scroll_channels(-1)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 678, in scroll_channels
    select_channel(new_selected_channel)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 651, in select_channel
    draw_messages_window(True)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 566, in draw_messages_window
    messages_pad.addstr(row, 1, line, color)
ValueError: embedded null character

is there a way to get around it?
maybe resetting the chat database (sadly losing all the conversations) or is there anything else I could try to make it work again?
thank you.

pdxlocations, 2025-07-14:
Could be related to #186  which was fixed in #192.  I've just issues a release with this fix so try updating to 1.3.15 and see if that helps.

Also, if you ever want to test clearing your message/node database, you can make a backup copy of client.db prior.

alestanga, 2025-07-15:
thank you, I've tried updating it to 1.3.15 but it still crashes:
Traceback (most recent call last):
  File "/home/radio/.local/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 1262, in _handleFromRadio
    fromRadio.ParseFromString(fromRadioBytes)
google.protobuf.message.DecodeError: Error parsing message with type 'meshtastic.protobuf.FromRadio'
Traceback (most recent call last):
  File "/home/radio/.local/lib/python3.9/site-packages/meshtastic/stream_interface.py", line 193, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/home/radio/.local/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 1268, in _handleFromRadio
    raise ex
  File "/home/radio/.local/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 1262, in _handleFromRadio
    fromRadio.ParseFromString(fromRadioBytes)
google.protobuf.message.DecodeError: Error parsing message with type 'meshtastic.protobuf.FromRadio'

2025-07-15 17:31:15,602 - ERROR - Fatal error: embedded null character
2025-07-15 17:31:15,606 - ERROR - Traceback: Traceback (most recent call last):
  File "/home/radio/.local/lib/python3.9/site-packages/contact/__main__.py", line 128, in start
    curses.wrapper(main)
  File "/usr/lib/python3.9/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/__main__.py", line 105, in main
    main_ui(stdscr)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 119, in main_ui
    handle_up()
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 184, in handle_up
    scroll_channels(-1)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 678, in scroll_channels
    select_channel(new_selected_channel)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 651, in select_channel
    draw_messages_window(True)
  File "/home/radio/.local/lib/python3.9/site-packages/contact/ui/contact_ui.py", line 566, in draw_messages_window
    messages_pad.addstr(row, 1, line, color)
ValueError: embedded null character

pdxlocations, 2025-07-15:
OK, thank you for testing that out!  There may be bad character/emoji or something in your MediumFast chat history that causes the crash.  If you can find a sql db viewer/editor, you could try removing or editing the last MediumFast entry.

I'd also be curious to know if starting with a fresh db solves your issue (for now).  You can just rename the client.db and the app should create a new one.

If we can narrow down the cause, I can try to reproduce the crash and hopefully fix this.  Thanks again for the reports!

alestanga, 2025-07-19:
I was on a work trip with but now I’m back.
So, I’ve made a copy of the DB, if you want to take a look at it.
Anyway, there were like 20 Messages or so with BLOB as content (I’ve used sqlitebrowser).
Once I’ve removed them it started working again.

pdxlocations, 2025-07-19:
BLOB!  OK, i wonder why that was crashing the app.  Do you have a copy of the corrupt db that you can send me to investigate?  Only if there's no sensitive messages in there of course.

closed 2025-07-21 by pdxlocations