pdxlocations/contact #205: [Feature Request] Add prompt and handling for new meshtastic firmware 2 second text message rate limit

rfschmid, 2025-07-14:
Meshtastic firmwares newer than 2.7.0 implement a rate limit to outgoing text messages. See https://github.com/meshtastic/firmware/pull/7139 . 

Newer Android client versions add an error message when trying to send a message too quickly, letting the user know they need to wait (similar to the traceroute rate limit.) Since typing multiple messages quickly is relatively easier with a computer keyboard, it seems like this limit will be easier to hit with this client than with others such as mobile device clients. Contact should at least tell the user if they're trying to send a message too quickly. We may also want to consider not showing the failed message in the outgoing messages window, or leaving the user's entered text in the message input window so they can more easily send their message after waiting sufficient time and not need to retype or copy/paste the message again. We could also consider automatically queue up the outgoing message and only try to send it after an appropriate delay, along with some UI indicator for showing the message waiting to be sent.

pdxlocations, 2025-07-14:
I think the firmware sends the errors to the clients for these things.  Catching and handling those would be great too!

pdxlocations, 2025-07-29:
Fixed in #212 by just checking the time between sends.