pdxlocations/contact #210: No length checking on Fixed PIN input

ai6yr, 2025-07-23:
Under Main Menu > Radio Settings > Bluetooth, Fixed PIN must be 6 digits; however, there is no length checking on this number, which can be much longer than 6 digits.  It may be that none of the inputs do any validation checking, which could result in invalid configurations and much frustration for users.

Changes are probably needed in settings_menu in the file control_ui.py to implement validation checking for this and other settings.

pdxlocations, 2025-07-25:
So true...

pdxlocations, 2025-07-25:
Work has begun on this issue in https://github.com/pdxlocations/contact/tree/input-validation

pdxlocations, 2025-07-27:
The framework is merged in #211.  Fixed Pin is "fixed" :-). New rules can be added to https://github.com/pdxlocations/contact/blob/main/contact/utilities/validation_rules.py

Thanks for the nudge, I never get around to validating inputs!